cleanflight/cleanflight #3090: Allow blackbox logging with disarmed FC (configurable?)

biasedlogic, 2020-03-14:
**Is your feature request related to a problem? Please describe.**

I'm using a flight controller (CC3D right now, but I'm going to update it to something with a magnetometer and newer µC soon) as a cheap logging IMU unit to analyze driving dynamics of vehicles. I know it's not the intended use, but the hardware is such a perfect fit for this purpose, that it's pity to reinvent the wheel.

Now, as far as I got this, I need to actually ARM the FC for the blackbox logging to start. I used an external chip (an Arduino) that pretends to be a cPPM RX and works the arming sequence upon powerup. This every now and then fails, because either the FC wasn't ready yet, or the GPS was late acquiring signal or the car wasn't level enough or for whatever of the 'thousand' reasons listed under https://github.com/betaflight/betaflight/wiki/Arming-Sequence-&-Safety

**Describe the solution you'd like**

I'd love to have a configuration switch that if set, would start blackbox logging of all the sensor data as well as calculated FC attitude and altitude upon power up and keep that logging up for as long there's power applied. A solution where a simple switch (say, one of the digital inputs on FC) would either force arming or trigger blackbox logging would also do the trick for me. Likely, even disabling all arming safety would suffice, as I don't mind the external Arduino, I just need it to work reliably 'blind', i.e. without user interaction beyond flipping the start switch.

**Describe alternatives you've considered**

Right now I emulate a receiver using an external arduino, but it doesn't reliably arm the FC.

I think I could hack a solution in myself, but I don't have any of the necessary toolchain set up, nor did I ever write anything for this µC architecture, so I hope this could be achieved with a simple enough hack that someone would be willing to do it for me... prettyplease ;-)

mikeller, 2020-03-14:
@biasedlogic: In Betaflight this can be done with `set blackbox_mode = ALWAYS`. Not sure if this feature ever made it into cleanflight.

biasedlogic, 2020-03-14:
Thank you, I will check it tomorrow!

Worst case I'll have to get an FC supported by Betaflight, but it's another can of worms with allegedly too fast data stream for open log to handle...

biasedlogic, 2020-03-15:
Oookay, good news and bad news ;-)
Good news: the setting you mentioned works and does what I need, BUT it has been introduced in some more recent version of Cleanflight. AND it was more recent, than dropping support of MAG sensor for my most used boards (Naze32 10DOF, F1 based board). In other words: it works, but I don't get MAG support, which I need. 

I guess I won't get around compiling my own firmware flavour.

mikeller, 2020-03-15:
@biasedlogic:

> I guess I won't get around compiling my own firmware flavour.

Looks like it - or get yourself a non-deprecated flight controller with an F4 MCU - if you go for one with a decent sized on-board flash chip or an SD card slot you'll also get around the problem of logging speed.

biasedlogic, 2020-03-15:
@mikeller yeah.. I bit the bullet and got a build toolchain installed, managed to get the SW built with MAG enabled (I kicked out a bunch of RX protocols and VTX features to get some space) and it works. I still have the limitation of not enough serial ports, better HW is still the way to solve it all, but no matter the board: on-board flash likely won't cut it, I need to log several hours of data (up to 24h), 10-100Hz is enough, but still it's going to be tons of data.

closed 2020-03-15 by biasedlogic