cleanflight/cleanflight #3091: Y6 and Octo X8 cannot be configured on SP F3 since release 2.1.

LasseTS, 2020-04-10:
Hi,

Y6 and Octo X8 cannot be configured on SP F3 since version 2.1 
It works in version 2.0.5 and earlier.

1.
Reset settings.
2.
Choose Y6 or Octo X8 under Configuration and Save-Reboot.
3.
After the SP F3 reboot the configuration defaults to Custom-Mixer.

Thanks,
Lasse

mikeller, 2020-04-11:
@LasseTS: Yes, 'exotic' pre-defined mixers have been removed from the firmware on F3 targets in order to make it fit into the available flash space.

Workarounds:
- upgrade your flight controller to an F4 or F7;
- use a custom mixer and manually add the mixer settings for the configuration you want (https://github.com/betaflight/betaflight/blob/master/docs/Mixer.md#custom-motor-mixing);
- stay on the older version of the firmware.

hydra, 2020-05-17:
Closing as inactive.

closed 2020-05-17 by hydra