cleanflight/cleanflight #3104: I lost one week to have servos working sp racing f3  with any result

kazmod1, 2020-06-19:
Like  in title. I have really work very hardly one week to have the servos, motor working in flying_wing with no result.  I have to sp racing f3 controller, both working in the same manner, with out servos.
I'm very curious, very is the error.  I really ask you to help me.

$M>'!$M>nø
CLI

# diff
# Cleanflight / SPRACINGF3 (SRF3) 2.5.0 Oct  1 2018 / 04:29:34 (08348b705) MSP API: 1.40

name Nowy Sp wing
resource MOTOR 3 NONE
resource MOTOR 4 NONE
resource SERVO 3 A11
resource SERVO 4 A12
mixer FLYING_WING
servo 4 1000 2000 1500 100 0
servo 5 1000 2000 1500 100 1

feature -RX_PPM
feature -TELEMETRY
feature RX_PARALLEL_PWM
map TAER1234
aux 0 0 0 1775 2000 0 0
set acc_calibration = -110,22,217
set use_unsynced_pwm = ON
set motor_pwm_protocol = OFF
set motor_pwm_rate = 200
set failsafe_procedure = AUTO-LAND
profile 0

rateprofile 0

# dump
# Cleanflight / SPRACINGF3 (SRF3) 2.5.0 Oct  1 2018 / 04:29:34 (08348b705) MSP API: 1.40

name Nowy Sp wing
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 NONE
resource MOTOR 4 NONE
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 A11
resource SERVO 4 A12
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource SONAR_TRIGGER 1 B00
resource SONAR_ECHO 1 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
[... 631 lines cut ...]
set f_yaw = 60
set p_level = 50
set i_level = 50
set d_level = 75
set level_limit = 55
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650
set throttle_limit_type = OFF
set throttle_limit_percent = 100

haslinghuis, 2020-06-22:
Why aren't you using servo `1 and 2` instead ‽ Now you are redirecting resources to servo `3 and 4` and are using servo `4 and 5` in your mixer?
```
resource SERVO 3 A11
resource SERVO 4 A12
mixer FLYING_WING
servo 4 1000 2000 1500 100 0
servo 5 1000 2000 1500 100 1
```

hydra, 2020-07-01:
Hi @kazmod1 a couple of points:
1) this is for an old version of Cleanflight., please use latest firmware.  Compile a custom version if you need to.
2) for support/configuration/setup/help please use Slack, RC Groups or Facebook user groups.

@haslinghuis thanks for helping out.

closed 2020-07-01 by hydra