cleanflight/cleanflight #3113: Rangefinder HC-SR04 problem when building firmware v4.2.0 for naze32

rafaelmaeuer, 2020-09-13:
**Describe the bug**

I try to build the latest release of cleanflight firmware (v4.2.0) for a Naze32 6DOF board to make use of a HC-SR04 sonar. Therefore I disabled some unused telemetry protocols to get the necessary flash space:

```
#undef USE_TELEMETRY_FRSKY
#undef USE_TELEMETRY_HOTT
#undef USE_TELEMETRY_LTM
#undef USE_TELEMETRY_SMARTPORT
#undef USE_TELEMETRY_FRSKY_HUB
```

and re-enabled the dropped code for the rangefinder hcsr04:

```
#define USE_RANGEFINDER
#define USE_RANGEFINDER_HCSR04
#define RANGEFINDER_HCSR04_TRIGGER_PIN       PB0
#define RANGEFINDER_HCSR04_ECHO_PIN          PB1
```

After building and flashing the firmware I enable the sonar on configuration tab in cleanflight configurator. Receiver is set to `PPM RX input` per default. After reboot the sonar is not recognized. The wiring works, as I tested the same setup with v1.13.0 of cleanflight firmware where the sonar works perfectly.

I did some further investigations and in `rangefinder.c` I found following code (line 99):

```
if (hcsr04Detect(dev, sonarConfig())) {   // FIXME: Do actual detection if HC-SR04 is plugged in
```

This FIXME gives me the hint that what I am trying to do isn't currently working???

**To Reproduce**

0. Connect HC-SR04 to Naze32 on Pins RC7/8: [How-To](https://quadmeup.com/connecting-ultrasonic-rangefinder-sonar-to-cleanflight/)
1. Checkout master branch, commit `HEAD CLFL_v4.2.0`
2. Apply above modifications in `cleanflight/src/main/target/NAZE/target.h`
3. Build the firmware
4. Flash firmware on Naze32 6DOF board
5. Connect with Cleanflight Configurator (v2.6.0)
6. Go to configuration tab
7. Enable `SONAR` feature
8. Save and reboot
9. Check if sonar is recognized

**Expected behavior**

After performing all steps described above, the sonar should be recognized and working as it is with firmware v1.13.0.

**Flight controller configuration**

```
# Cleanflight / NAZE (AFNA) 4.2.0 Sep 13 2020 / 20:39:07 (norevision) MSP API: 1.43
batch start
feature -DYNAMIC_FILTER
feature RANGEFINDER
set motor_pwm_protocol = PWM
set pid_process_denom = 16

profile 0

rateprofile 0
batch end
```

```
# resource show all
IO
A00: PPM
A01: FREE
A02: FREE
A03: FREE
A04: ADC_BATT
A05: ADC_EXT
A06: FREE
A07: FREE
A08: MOTOR 1
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: MOTOR 2
A12: BEEPER
A13: SWD
A14: SWD
A15: FREE
B00: FREE
B01: FREE
B02: INVERTER 2
B03: LED 1
B04: LED 2
B05: FREE
B06: MOTOR 3
B07: MOTOR 4
B08: FREE
B09: FREE
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: GYRO_EXTI
C14: FREE
C15: FREE

DMA:
DMA1 Channel 1: ADC
DMA1 Channel 2: FREE
DMA1 Channel 3: FREE
DMA1 Channel 4: SERIAL_TX 1
DMA1 Channel 5: SERIAL_RX 1
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
```

**Setup / Versions**
 - Flight controller;
    - Naze32 6DOF (Afroflight) bought on eBay
 - Other components;
    - HC-SR04 bought on http://www.icstation.com
    - External power supply 5V
 - how are the different components wired up.
    - Power 5V <-> HC-SR04 VCC
    - Power GND <-> HC-SR04 GND <-> Naze32 GND
    - Naze32 RC7 <-> 1K Resistor <-> HC-SR04 Trig
    - Naze32 RC8 <-> 1K Resistor <-> HC-SR04 Echo

**Additional context**

I also tried it using the `RANGEFINDER_HCSR04I2C`, but when connecting the HC-SR04 to the I2C ports of the Naze32 the gyro isn't detected anymore...

Another try was disabling software-serial-2 as it shares the same pins as HC-SR04 when not using PWM-Mode - without success...

Any help is appreciated

rafaelmaeuer, 2020-10-13:
No idea anyone???

haslinghuis, 2020-10-14:
// Rangefinder dropped to make flash space

haslinghuis, 2020-10-14:
src/main/target/NAZE/target.h:// Rangefinder dropped to make flash space

rafaelmaeuer, 2020-10-17:
I know it was dropped to make flash space for naze32, but does it work in current firmware with other boards?

haslinghuis, 2020-10-19:
Drivers should work but only produces altitude data afaik and it's not part of unified targets. Only used it on arduino so far.

rafaelmaeuer, 2020-10-20:
The firmware compiles fine and the rangefinder option is selectable, but the hc-sr04 sensor isn't detected. I use the same wiring as with firmware v1.13.0 where it is recognized perfectly and shows correct output data.

haslinghuis, 2020-10-21:
https://github.com/betaflight/betaflight/pull/9873

closed 2020-11-29 by stale[bot]