cleanflight/cleanflight #3115: Unlable to connect the configurator to STM32F411CEU6 after flashing

HristoB, 2020-09-19:
**Is your feature request related to a problem? Please describe.**
Hi, I've a couple of STM32F411CEU6 development boards and so far I've managed to build and flash a STM32F411 target via the USB port in DFU mode as well as with SWD with STLinkV2 following the documentation here. The problem I have is that I cannot connect the board to CF-configurator via a UART as well as the USB. Unfortunately, my experience and knowledge with coding is only with Arduino IDE and would be great if anyone could suggest a solution. 

**Describe the solution you'd like**
Enable a UART for possible board configuration via FTDI

**Additional context**
I've experimented before with CF on F1 boards and could flash and connect the configurator to UART1 via FTDI with a NAZE target and build and fly my home made FC. 
Please note that feature requests are not 'fire and forget'. It is a lot more likely that the feature you would like to have will be implemented if you keep watching your feature request, and provide more details to developers looking into implementing your feature, and help them with testing.

haslinghuis, 2020-09-22:
See https://oscarliang.com/connect-fc-broken-usb/

HristoB, 2020-09-22:
> See https://oscarliang.com/connect-fc-broken-usb/

Hi, thanks for the reply. 
I've done that already. Same thing. Checked target.h for a couple of F411 targets, MSP is enabled on UART1 and connected to the appropriate pins according the TX and RX assignment. Still cant connect. Flashed and connected to my F1 without problems after flashing. In STMCube I can connect to UART1 PA9 and PA10) and erase and flash but not in the  configurator. Flashed FLYWOOF411 target as it's using the same chip and using PB6 and PB7 as UART1, same story. The F1 Naze clone I've built on a "el cheapo" blue pill works fine on a perf board :-) with BMP180 barometer and MPU6050, no problems from CF 1.10 to 2.2.0...
Unfortunately my coding is very basic and only on Arduino...

HristoB, 2020-09-23:
> See https://oscarliang.com/connect-fc-broken-usb/

Hi, 
I think I've found out what's going on here after a bit of digging around.
I've erased the chip via stlink and flashed with CF 2.4 generic STM32F411 target hex but still could not connect. Then started it in DFU mode and flashed via the usb port and after reboot the ttyACM appeared and managed to connect. The problem was that there were no UARTs listed in Ports tab so went to CLI and configured th PA9 and PA10 as UART1 TX and RX with "resource SERIAL_TX 1 A09 and SERIAL_RX 1 A10", enabled MSP on UART1 and then I could connect with FTDI. Enabled the built-in LEDs as well. Now what's left to be done is to pick a F411 target, matek or flywoo, whatever, and set the pin mapping as they're using the same chip. 
I'm wondering if I could do that in the target.h file and compile the hex..
Hope this helps other noobs like myself, trying to build their own FCs

haslinghuis, 2020-09-23:
Maybe you could craft the STMI unified target for the STM32F411CEU6 providing the target.h file you mentioned.

HristoB, 2020-09-24:
> Maybe you could craft the STMI unified target for the STM32F411CEU6 providing the target.h file you mentioned.

Easily said than done :-D The closest I got to coding was writing sketches in Arduino IDE but it's not a reason to give up. I've admit, with this one I took a bite larger than I can chew, but hey, need to try new things and challenge yourself. 

Still going through the CF documentation but will get there. I'm in a process of getting the schematic ready but the thing is the chinese board dont give access to all of the chip pins so have to work out the pin assignment based on the available boards with the same chip. 

Will work it out eventually and wouldnt mind if someone drops a hint here and there :-)

closed 2020-11-01 by stale[bot]